alexforencich/verilog-axi #1: Quick question about clock domain crossing (CDC)

olagrottvik, 2019-06-26:
Hi Alex,

Do you plan to add support for "auto" clock domain crossing between master and slaves? E.g. I have multiple slaves that operate in different domains of each other. Now, all my writing and reading of register in the slave needs to go through CDC as the AXI bus is operating in another domain. However, if the bus itself were CDC'd for each slave in another domain than the master, a lot of hassle could be avoided. I also think I would save a lot of resources.

I understand completely if this is far fetched :)

Ola

alexforencich, 2019-06-26:
For AXI or for AXI lite?

A CDC module is certainly something I have thought about implementing. The only problem is that it isn't exactly trivial. The main problem I think is not so much the multiple clock domains, but the multiple reset domains. Most AXI masters are not tolerant of transactions getting "lost," so something that crosses into a different reset domain has to be able to make sure that in-flight transactions get handled correctly across a reset of only one side or the other. This makes the required logic significantly more complex, at least for full AXI. For AXI lite, only one transaction in flight at a time is supported, there are fewer trade-offs to consider (area vs. latency vs. throughput).

olagrottvik, 2019-06-26:
For me, only AXI lite is needed.

alexforencich, 2019-06-26:
Alright, I'll look in to putting some sort of CDC module for AXI lite.  Might be a little while before I have time to work on it, though.

olagrottvik, 2019-06-26:
Cool, thanks!

alexforencich, 2019-06-26:
Try this.  It's a bit slow - two or three cycles of latency in each direction for every operation - but I think it will work.  Works in sim, but I don't have any FPGA projects that need it right now so I can't check the timing constraints.  There is a high probability I have some signal names wrong.  

[axil_cdc.zip](https://github.com/alexforencich/verilog-axi/files/3332091/axil_cdc.zip)

olagrottvik, 2019-07-02:
Thanks a million! I'll have to try it after the summer holidays.

alexforencich, 2019-07-09:
So, as expected, the TCL constraints file had some issues.  So I fixed that, but then realized Vivado had merged some of the registers into a block RAM as output registers in the test design I was working with, so I had to figure out how to get the timing constraints TCL file to track down the correct instances by following the nets.  However, I think it will work now, so I went ahead and committed it.  Let me know how it works for you.

olagrottvik, 2019-07-09:
I don't see which files are supposed to be the cdc-module and also no TCL file is in the repo. Did you push it?

alexforencich, 2019-07-09:
Whoops; it's up there now

olagrottvik, 2019-08-14:
After a quick test, it seems like it works fine. I'll do a proper stress test later, but I'll close this for now.

Thanks a lot again!

closed 2019-08-14 by olagrottvik