alexforencich/verilog-axi #13: Long logic depth leads to bad timing

luyong6, 2020-11-15:
Hi another question (and also curious): 

Why dma_wr uses a different structure to dma_rd on the generation of status_len/tag? 
dma_wr uses status_fifo, but dma_rd doesn't. 

My recent synthesis tells me that it has 16 logic levels. It's a challenge for 300MHz+. 

```
Max Delay Paths
--------------------------------------------------------------------------------------
Slack (VIOLATED) :        -1.535ns  (required time - arrival time)
  Source:                 u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/offset_reg_reg[1]/C
                            (rising edge-triggered cell FDRE clocked by txoutclk_out[0]  {rise@0.000ns fall@1.552ns period=3.103ns})
  Destination:            u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/status_fifo_len_reg_0_31_0_13/RAMF_D1/I
                            (rising edge-triggered cell RAMD32 clocked by txoutclk_out[0]  {rise@0.000ns fall@1.552ns period=3.103ns})
  Path Group:             txoutclk_out[0]
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            3.103ns  (txoutclk_out[0] rise@3.103ns - txoutclk_out[0] rise@0.000ns)
  Data Path Delay:        4.406ns  (logic 1.681ns (38.153%)  route 2.725ns (61.848%))
  Logic Levels:           16  (CARRY8=2 LUT2=1 LUT3=1 LUT4=2 LUT5=3 LUT6=7)
  Clock Path Skew:        -0.105ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    3.522ns = ( 6.625 - 3.103 )
    Source Clock Delay      (SCD):    3.931ns
    Clock Pessimism Removal (CPR):    0.304ns
  Clock Uncertainty:      0.046ns  ((TSJ^2 + DJ^2)^1/2) / 2 + PE
    Total System Jitter     (TSJ):    0.071ns
    Discrete Jitter          (DJ):    0.060ns
    Phase Error              (PE):    0.000ns
  Clock Net Delay (Source):      3.716ns (routing 1.706ns, distribution 2.010ns)
  Clock Net Delay (Destination): 3.330ns (routing 1.550ns, distribution 1.780ns)
```

dma_rd.v only generates status_tag output. I used the same way to generate status_len (this is important to count how many bytes has been transferred) and it looks much simpler than the status_fifo in dma_wr.v

alexforencich, 2020-11-18:
It uses a different structure because the ordering requirements are different.  For the read DMA, no reordering is allowed at all, so the ARID signal is fixed at 0.  The downside is that this has the potential to cause issues in AXI interconnects as the interconnect must preserve the ordering, and usually this is done by blocking transfers to multiple devices using the same ID, so this can possibly result in a performance degredation through the AXI interconnect.  For the write DMA, reordering is no problem at all, so it drives different values on AWID to get higher throughput through the interconnect, but then has to deal with write responses coming back out-of-order.  Now, it is, in principle, possible to deal with reordering on the read side by adding a buffer to the DMA engine, but this increases complexity, resource consumption, and (most likely) latency.  

16 logic levels seems excessive, can you post the full timing path?  There may be some simple changes I can try to break the path.

luyong6, 2020-11-19:
BTW: I have reduced the FIFO depth. So the reports shows 1 logic level less. 
```
parameter STATUS_FIFO_ADDR_WIDTH = 4; //Try to optimize timing, changed from 5 to 4.
```

```
--------------------------------------------------------------------------------------
Slack (VIOLATED) :        -1.423ns  (required time - arrival time)
  Source:                 u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/offset_reg_reg[4]/C
                            (rising edge-triggered cell FDRE clocked by txoutclk_out[0]  {rise@0.000ns fall@1.552ns period=3.103ns})
  Destination:            u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/status_fifo_len_reg_0_15_0_13/RAMG/I
                            (rising edge-triggered cell RAMD32 clocked by txoutclk_out[0]  {rise@0.000ns fall@1.552ns period=3.103ns})
  Path Group:             txoutclk_out[0]
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            3.103ns  (txoutclk_out[0] rise@3.103ns - txoutclk_out[0] rise@0.000ns)
  Data Path Delay:        4.294ns  (logic 1.445ns (33.652%)  route 2.849ns (66.348%))
  Logic Levels:           15  (CARRY8=2 LUT4=2 LUT6=11)
  Clock Path Skew:        -0.129ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    2.895ns = ( 5.998 - 3.103 ) 
    Source Clock Delay      (SCD):    3.269ns
    Clock Pessimism Removal (CPR):    0.245ns
  Clock Uncertainty:      0.046ns  ((TSJ^2 + DJ^2)^1/2) / 2 + PE
    Total System Jitter     (TSJ):    0.071ns
    Discrete Jitter          (DJ):    0.060ns
    Phase Error              (PE):    0.000ns
  Clock Net Delay (Source):      3.054ns (routing 1.497ns, distribution 1.557ns)
  Clock Net Delay (Destination): 2.703ns (routing 1.361ns, distribution 1.342ns)

    Location             Delay type                Incr(ns)  Path(ns)    Netlist Resource(s)
  -------------------------------------------------------------------    -------------------
                         (clock txoutclk_out[0] rise edge)
                                                      0.000     0.000 r  
    GTYE4_CHANNEL_X0Y8   GTYE4_CHANNEL                0.000     0.000 r  u_cmac/qsfp_cmac_inst/inst/cmac_usplus_0_gt_i/inst/gen_gtwizard_gtye4_top.cmac_usplus_0_gt_gtwizard_gtye4_inst/gen_gtwizard_gtye4.gen_channel_container[2].gen_enabled_channel.gtye4_channel_wrapper_inst/channel_inst/gtye4_channel_gen.gen_gtye4_channel_inst[0].GTYE4_CHANNEL_PRIM_INST/TXOUTCLK
                         net (fo=2, routed)           0.085     0.085    u_cmac/qsfp_cmac_inst/inst/cmac_gtwiz_userclk_tx_inst/txoutclk_out[0]
    BUFG_GT_X0Y70        BUFG_GT (Prop_BUFG_GT_I_O)
                                                      0.130     0.215 r  u_cmac/qsfp_cmac_inst/inst/cmac_gtwiz_userclk_tx_inst/gen_gtwiz_userclk_tx_main.bufg_gt_usrclk_inst/O
    X3Y4 (CLOCK_ROOT)    net (fo=47258, routed)       3.054     3.269    u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/cmac_clk
    SLICE_X87Y219        FDRE                                         r  u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/offset_reg_reg[4]/C
  -------------------------------------------------------------------    -------------------
    SLICE_X87Y219        FDRE (Prop_BFF_SLICEM_C_Q)
                                                      0.078     3.347 f  u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/offset_reg_reg[4]/Q
                         net (fo=46, routed)          0.277     3.624    u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/offset_reg[4]
    SLICE_X87Y217        LUT6 (Prop_G6LUT_SLICEM_I0_O)
                                                      0.150     3.774 r  u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/m_axi_wstrb_reg[3]_i_19/O
                         net (fo=71, routed)          0.380     4.154    u_fpga/u_net_wrapper/u_net_core/udp_payload_rx_fifo/temp_m_axi_wstrb_reg[60]_i_6_1
    SLICE_X90Y218        LUT6 (Prop_H6LUT_SLICEL_I1_O)
                                                      0.090     4.244 r  u_fpga/u_net_wrapper/u_net_core/udp_payload_rx_fifo/temp_m_axi_wstrb_reg[58]_i_10/O
                         net (fo=3, routed)           0.181     4.425    u_fpga/u_net_wrapper/u_net_core/udp_payload_rx_fifo/mem_reg_7_15
    SLICE_X91Y217        LUT6 (Prop_H6LUT_SLICEM_I1_O)
                                                      0.152     4.577 f  u_fpga/u_net_wrapper/u_net_core/udp_payload_rx_fifo/temp_m_axi_wstrb_reg[58]_i_9/O
                         net (fo=4, routed)           0.276     4.853    u_fpga/u_net_wrapper/u_net_core/udp_payload_rx_fifo/mem_reg_7_4
    SLICE_X92Y216        LUT6 (Prop_H6LUT_SLICEL_I0_O)
                                                      0.036     4.889 f  u_fpga/u_net_wrapper/u_net_core/udp_payload_rx_fifo/temp_m_axi_wstrb_reg[54]_i_7/O
                         net (fo=2, routed)           0.184     5.073    u_fpga/u_net_wrapper/u_net_core/udp_payload_rx_fifo/offset_reg_reg[0]_rep__1_8
    SLICE_X93Y217        LUT6 (Prop_H6LUT_SLICEL_I1_O)
                                                      0.124     5.197 f  u_fpga/u_net_wrapper/u_net_core/udp_payload_rx_fifo/temp_m_axi_wstrb_reg[54]_i_3_comp/O
                         net (fo=6, routed)           0.150     5.347    u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/temp_m_axi_wstrb_reg_reg[54]_0
    SLICE_X93Y213        LUT4 (Prop_F6LUT_SLICEL_I0_O)
[... cut ...]
                                                     -0.056     6.140    u_fpga/u_net_wrapper/u_udp_aximm_write/axi_dma_wr_inst/status_fifo_len_reg_0_15_0_13/RAMG
  -------------------------------------------------------------------
                         required time                          6.140    
                         arrival time                          -7.563    
  -------------------------------------------------------------------
                         slack                                 -1.423    
```

luyong6, 2020-11-19:
BTW, I only use AWID = 0.

alexforencich, 2020-11-19:
Actually, looking at the code again, my initial description was partially incorrect.  Both modules use ID = 0.  However, the DMA write module should not indicate that the operation is complete until all of the AXI write responses come back, otherwise you could get into a situation where you are operating on data that is not completely written (ask me how I know that....).  So, the response data is stored in a queue and held until the write response comes back so that when the DMA module reports that the operation is complete, all of the data on the AXI side will be consistent.  

I'm thinking maybe the solution is to insert a register stage before writing into the FIFO, and possibly push some logic over that register.  There is a lot of stuff that affects the length field that gets placed in the FIFO, after all.

KireinaHoro, 2024-01-05:
Sorry for reviving an old issue, but I'm also struggling with this at 250 MHz on a VU9P with the same 16-stage path:

```
Max Delay Paths
--------------------------------------------------------------------------------------
Slack (VIOLATED) :        -0.187ns  (required time - arrival time)
  Source:                 design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/offset_reg_reg[2]_replica/C
                            (rising edge-triggered cell FDRE clocked by clk_out1_design_1_clk_wiz_0_0  {rise@0.000ns fall@2.000ns period=4.000ns})
  Destination:            design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/status_fifo_len_reg_0_31_14_15/RAMA_D1/I
                            (rising edge-triggered cell RAMD32 clocked by clk_out1_design_1_clk_wiz_0_0  {rise@0.000ns fall@2.000ns period=4.000ns})
  Path Group:             clk_out1_design_1_clk_wiz_0_0
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            4.000ns  (clk_out1_design_1_clk_wiz_0_0 rise@4.000ns - clk_out1_design_1_clk_wiz_0_0 rise@0.000ns)
  Data Path Delay:        4.166ns  (logic 1.301ns (31.231%)  route 2.865ns (68.769%))
  Logic Levels:           15  (CARRY8=2 LUT4=4 LUT5=1 LUT6=8)
  Clock Path Skew:        0.106ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    4.711ns = ( 8.711 - 4.000 )
    Source Clock Delay      (SCD):    4.283ns
    Clock Pessimism Removal (CPR):    -0.322ns
  Clock Uncertainty:      0.061ns  ((TSJ^2 + DJ^2)^1/2) / 2 + PE
    Total System Jitter     (TSJ):    0.071ns
    Discrete Jitter          (DJ):    0.098ns
    Phase Error              (PE):    0.000ns
  Clock Net Delay (Source):      3.244ns (routing 1.984ns, distribution 1.260ns)
  Clock Net Delay (Destination): 3.133ns (routing 1.804ns, distribution 1.329ns)

    Location             Delay type                Incr(ns)  Path(ns)    Netlist Resource(s)
  -------------------------------------------------------------------    -------------------
                         (clock clk_out1_design_1_clk_wiz_0_0 rise edge)
                                                      0.000     0.000 r
    AY26                                              0.000     0.000 r  pgrc0_clk_clk_p (IN)
                         net (fo=0)                   0.000     0.000    design_1_i/clk_wiz_0/inst/clkin1_ibufds/I
    HPIOBDIFFINBUF_X1Y154
                         DIFFINBUF (Prop_DIFFINBUF_HPIOBDIFFINBUF_DIFF_IN_P_O)
                                                      0.474     0.474 r  design_1_i/clk_wiz_0/inst/clkin1_ibufds/DIFFINBUF_INST/O
                         net (fo=1, routed)           0.050     0.524    design_1_i/clk_wiz_0/inst/clkin1_ibufds/OUT
    AY26                 IBUFCTRL (Prop_IBUFCTRL_HPIOB_M_I_O)
                                                      0.000     0.524 r  design_1_i/clk_wiz_0/inst/clkin1_ibufds/IBUFCTRL_INST/O
                         net (fo=1, routed)           0.375     0.899    design_1_i/clk_wiz_0/inst/clk_in1_design_1_clk_wiz_0_0
    MMCM_X1Y6            MMCME4_ADV (Prop_MMCM_CLKIN1_CLKOUT0)
                                                     -0.127     0.772 r  design_1_i/clk_wiz_0/inst/mmcme4_adv_inst/CLKOUT0
                         net (fo=1, routed)           0.239     1.011    design_1_i/clk_wiz_0/inst/clk_out1_design_1_clk_wiz_0_0
    BUFGCE_X1Y167        BUFGCE (Prop_BUFCE_BUFGCE_I_O)
                                                      0.028     1.039 r  design_1_i/clk_wiz_0/inst/clkout1_buf/O
    X0Y3 (CLOCK_ROOT)    net (fo=74067, routed)       3.244     4.283    design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/clk
    SLR Crossing[1->0]
    SLICE_X11Y204        FDRE                                         r  design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/offset_reg_reg[2]_replica/C
  -------------------------------------------------------------------    -------------------
    SLICE_X11Y204        FDRE (Prop_FFF2_SLICEM_C_Q)
                                                      0.069     4.352 r  design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/offset_reg_reg[2]_replica/Q
                         net (fo=16, routed)          0.154     4.506    design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/I29[1]_repN
    SLICE_X11Y205        LUT6 (Prop_G6LUT_SLICEM_I4_O)
                                                      0.135     4.641 r  design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/shift_axis_extra_cycle_reg_i_115/O
                         net (fo=89, routed)          0.544     5.185    design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/offset_reg_reg[5]_3
    SLICE_X14Y205        LUT5 (Prop_G6LUT_SLICEL_I3_O)
                                                      0.114     5.299 r  design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/out_fifo_wstrb_reg_0_31_14_27_i_169/O
                         net (fo=4, routed)           0.171     5.470    design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/out_fifo_wstrb_reg_0_31_14_27_i_169_n_0
    SLICE_X14Y204        LUT6 (Prop_H6LUT_SLICEL_I2_O)
                                                      0.135     5.605 f  design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/out_fifo_wstrb_reg_0_31_28_41_i_111/O
                         net (fo=4, routed)           0.116     5.721    design_1_i/PioNicEngine_0/inst/axiDma/axi_dma_wr_inst/out_fifo_wstrb_reg_0_31_28_41_i_111_n_0
    SLICE_X14Y203        LUT6 (Prop_E6LUT_SLICEL_I1_O)
[... cut ...]
  -------------------------------------------------------------------
                         required time                          8.261
                         arrival time                          -8.449
  -------------------------------------------------------------------
                         slack                                 -0.187
```

I'm trying a bit of floorplanning, but it's quite tough.  Do you have plans to implement the changes that would register the input to the status FIFO?  Thanks a lot!

KireinaHoro, 2024-01-24:
@alexforencich Not sure if this is enough to revive the issue... Otherwise I'll recreate a new one