alexforencich/verilog-axi #14: axil_ram handshake between AW/AR-READY and B/R-RESP

jimmysitu, 2021-05-06:
Hi, @alexforencich 

I am doing a formal verificaion for verilog-axi IPs, see my repo [verilog-axi-formal](https://github.com/jimmysitu/verilog-axi-formal). And I found an AXI4 lite handshake issue with axil_ram interface, the aw/w-ready is assert with bvalid at the same clock, see the code below
https://github.com/alexforencich/verilog-axi/blob/314ea7dbf917f4e81da168cb20ce1bfa8ef64acf/rtl/axil_ram.v#L118-L124

And the AXI standard says,
> the slave must wait for both WVALID and WREADY to be asserted before asserting BVALID
An explain from https://community.arm.com/ is [here](https://community.arm.com/developer/ip-products/system/f/soc-design-forum/8113/axi-handshake-between-aw-ar-ready-and-b-r-resp)
Which means bvalid should at least one clock after wready.

alexforencich, 2021-05-07:
I guess you didn't see #11.  This is a known issue that's on my to-do list to fix; I haven't decided the best way to go about doing that yet.

jimmysitu, 2022-01-21:
Hi, @alexforencich , is there any update about this bug?

Bald-Badger, 2022-04-04:
The fix mentioned in #11 is legit. at least it passed my Cyclone V HPS AXI R/W test (and it didn't pass before the fix)