alexforencich/verilog-axi #16: axi_interconnect question

aignacio, 2021-05-28:
Hey @alexforencich,

about the axi_interonnect module, could you give an example of how to create the memory map of masters / slaves using your design? I'm bit confused of how to define it like exists for instance in the [bus matrix](https://developer.arm.com/documentation/ddi0479/b/Advanced-AHB-Components/Bus-matrix/Key-features) from ARM.  Basically in the bus matrix you have a xml where you define the relation between m+s and then generates the design of the interconnect, in your case I suppose we handle the parameters of the design, but how exactly I can tweak it to have a scenario like this:

```bash
  _______________
| 0x0000 | slave # 0 |
| 0x1000  | slave # 1 |
| 0x4000 | slave # 2 |
| 0x5000 | slave # 3 |
| 0x6000 | slave # 4 |
  ⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻
```

alexforencich, 2021-05-28:
You do this by setting `M_BASE_ADDR` and `M_ADDR_WIDTH`.  These parameters are formed by concatenating the values together.  For `M_BASE_ADDR`, this will be `M_COUNT * M_REGIONS * ADDR_WIDTH` bits in total.  For `M_ADDR_WIDTH`, this will be `M_COUNT * M_REGIONS * 32` bits in total.  

The routing is all done by masking the address, so all assignments have to be aligned based on the region size and all region sizes must be even powers of two.  Also, all assignments have to fall on 4 KB address boundaries.  

The interconnect module also has the ability to automatically assign addresses if you specify only `M_ADDR_WIDTH` and set `M_BASE_ADDR` to 0.  

So, for your table, slave # 1 is potentially problematic as it starts on 0x1000, which due to alignment can only extend to 0x1FFF.  But slave # 2 starts at 0x4000 instead of 0x2000.  So you will have a hole there from 0x2000 through 0x3FFF.  If you want to assign those addresses, you'll need to specify both `M_BASE_ADDR` and `M_ADDR_WIDTH` like so:

    .ADDR_WIDTH(16),
    .M_REGIONS(1),
    .M_BASE_ADDR({16'h6000, 16'h5000, 16'h4000, 16'h1000, 6'h0000}),
    .M_ADDR_WIDTH({5{32'd10})

Without the hole, you can simply do:

    .ADDR_WIDTH(16),
    .M_REGIONS(1),
    .M_BASE_ADDR(0),
    .M_ADDR_WIDTH({5{32'd10})

and the assigned addresses will be 0x0000, 0x1000, 0x2000, 0x3000, and 0x4000.  

The automatic address assignment also works with different region sizes.  The base address is aligned based on the region size, so you may end up with holes depending on the exact sizes and the sequence they appear in.  For example:

    .M_BASE_ADDR(0),
    .M_ADDR_WIDTH({32'd10, 32'd10, 32'd11})

will result in 0x0000-0x1FFF, 0x2000-0x2FFF, and 0x3000-0x3FFF.  But if you do

    .M_BASE_ADDR(0),
    .M_ADDR_WIDTH({32'd10, 32'd11, 32'd10})

then you'll get 0x0000-0x0FFF, 0x2000-0x3FFF, and 0x4000-0x4FFF, with a hole from 0x1000-0x1FFF.  In this case, you can manually assign addresses:

    .ADDR_WIDTH(16),
    .M_REGIONS(1),
    .M_BASE_ADDR({16'h1000, 16'h2000, 16'h0000}),
    .M_ADDR_WIDTH({32'd10, 32'd11, 32'd10})

This will result in 0x0000-0x0FFF, 0x2000-0x3FFF, and 0x1000-0x1FFF, with no holes.

aignacio, 2021-05-30:
Thanks for the reply!
just to see if I understood correctly, considering 2x masters (with both masters having the same memory map) and 5x slaves with the following scenario:

```verilog
  _________________________
| 0x0000-0x0FFF | slave # 0 |
| 0x1000-0x2FFF | slave # 1 |
| 0x4000-0x4FFF | slave # 2 |
| 0x5000-0x5FFF | slave # 3 |
| 0x6000-0x6FFF | slave # 4 |
  ⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻
```
```verilog
.ADDR_WIDTH(16),
.M_REGIONS(1),
.M_BASE_ADDR({16'h6000, 16'h5000, 16'h4000, 16'h1000, 6'h0000}),
.M_ADDR_WIDTH({32'd10, 32'd10, 32'd10, 32'd11, 32'd10})
```
is this configuration correct? I will have a 1KB bubble from the second slave but it's fine.

alexforencich, 2021-05-31:
The regions have to be aligned based on the size.  If you start at 0x1000, then that region must end at 0x1FFF.  But, if you start at 0x2000, then you can go to 0x3FFF.  So what you listed is not allowed, but you could do:

```verilog
  _________________________
| 0x0000-0x0FFF | slave # 0 |
| 0x2000-0x3FFF | slave # 1 |
| 0x4000-0x4FFF | slave # 2 |
| 0x5000-0x5FFF | slave # 3 |
| 0x6000-0x6FFF | slave # 4 |
  ⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻
```
```verilog
.S_COUNT(2),
.M_COUNT(5),
.ADDR_WIDTH(16),
.M_REGIONS(1),
.M_BASE_ADDR({16'h6000, 16'h5000, 16'h4000, 16'h2000, 6'h0000}),
.M_ADDR_WIDTH({32'd10, 32'd10, 32'd10, 32'd11, 32'd10})
```
Alternatively, you could split 0x1000-0x2FFF into two regions, 0x1000-0x1FFF and 0x2000-0x2FFF and then do something like this:

```verilog
.S_COUNT(2),
.M_COUNT(5),
.ADDR_WIDTH(16),
.M_REGIONS(2),
.M_BASE_ADDR({16'h0000, 16'h6000, 16'h0000, 16'h5000, 16'h0000, 16'h4000, 16'h3000, 16'h2000, 16'h0000, 16'h0000}),
.M_ADDR_WIDTH({32'd0, 32'd10, 32'd0, 32'd10, 32'd0, 32'd10, 32'd10, 32'd10, 32'd0, 32'd10})
```

aignacio, 2021-05-31:
tks again, I'm sorry but it's still not clear why exactly 0x1000 should end at 0x1FFF. I understand it should fit in 4KB boundaries, but 0x2000-0x3FFF is 8KB size and what I proposed it's more or less the same (0x1000-0x2FFF). What alignment exactly I'm missing? btw, what's the masking you said with `M_ADDR_WIDTH`?

alexforencich, 2021-05-31:
There are two ways to think about it.  

Way 1: the regions must be aligned.  This means that `BASE_ADDR % REGION_SIZE == 0`.  So `0x1000 % 0x1000` == 0, which is aligned.  But `0x1000 % 0x2000 == 0x1000`, which is not aligned.  

Way 2: the switch address routing works by generating a bit mask that covers the fixed MSBs.  So the mask could be 0xF000, 0xE000, 0xC000, or 0x8000 for a 16 bit address.  With a mask of 0xF000, the 4 MSBs must be fixed and determine the routing, but the lower 12 bits are passed through the interconnect.  So for a base address of 0x1000, you must use a mask of 0xF000, which means that you can only address 0x1000-0x1FFF in that region since the 4 MSBs are fixed.  If you want a region twice the size, then you must use a mask of 0xE000, which permits base addresses of 0x0000, 0x2000, 0x4000, 0x6000, 0x8000, etc.

aignacio, 2021-05-31:
I see, so based on your explanation I suppose the following is allowed.
```verilog
  _______________________________
| 0x2000-0x3FFF | slave # 0 - 8KB |
| 0x4000-0x5FFF | slave # 1 - 8KB |
| 0x6000-0x6FFF | slave # 2 - 4KB |
  ⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻⎻
```
```verilog
.S_COUNT(2),
.M_COUNT(3),
.ADDR_WIDTH(16),
.M_REGIONS(1),
.M_BASE_ADDR({16'h6000, 16'h4000, 6'h2000}),
.M_ADDR_WIDTH({32'd10, 32'd11, 32'd11})
```

alexforencich, 2021-05-31:
Yes, that should work fine.  Also, the module does internal checking for address range overlaps and other configuration issues at synthesis time and will stop synthesis and print out which entries are problematic.  So if you try to use 0x0000-0x0FFF and 0x1000-0x2FFF, this will be reported as an overlap as 0x1000 will get masked off to 0x0000, which then conflicts with the first range.

aignacio, 2021-06-01:
oh great, thanks again for the answers and the feedback!

closed 2021-06-01 by aignacio