alexforencich/verilog-axi #17: BRAM inference for Xilinx FPGAs

aignacio, 2021-06-24:
Hey @alexforencich,

for the `axi_ram.v` design, do you have a version with the macros instantiations (BRAM_SDP_MACRO/BRAM_TDP_MACRO) or a tip to force bram inference by Vivado? I'm actually suffering with Vivado consuming LUT RAMs instead of BRAMs for my memories in the design....I suppose only adding (* ram_style = "block" *) will not force the inference once the design doesn't follow "Xilinx pattern of design".

alexforencich, 2021-06-24:
No; I don't use macros because they're not portable.  Maybe I will have to revisit that at some point and add a bunch of conditionals to select the manufacturer- and/or device-specific macros.  Actually....maybe what I should do is make manufacturer- and/or device-specific wrappers, and then you can simply include the verilog file for the part you're using.  I actually quite like that idea, that way I don't need to deal with defines or threading through a bunch of parameters.  

The AXI RAMs should work fine for BRAM and URAM though.  What parameters are you specifying for the axi_ram instances when you're getting LUTRAMs instead of BRAMs?

aignacio, 2021-06-24:
It's seems a good idea having wrappers for different vendors, my current wrapper around the axi_ram is this one [here](https://raw.githubusercontent.com/aignacio/mpsoc_example/master/rtl/misc/axi_mem_wrapper.sv).
```Verilog
 localparam ADDR_RAM = $clog2(MEM_KB*1024);

  axi_ram #(
    // Width of data bus in bits
    .DATA_WIDTH(32),
    // Width of address bus in bits
    .ADDR_WIDTH(ADDR_RAM),
    // Width of ID signal
    .ID_WIDTH(1),
    // Extra pipeline register on output
    .PIPELINE_OUTPUT(1)
  ) u_ram (
```

alexforencich, 2021-06-24:
Hm, I would expect that to infer BRAMs.  Have you tried adding another pipeline stage or two?  And are you seeing any messages in the logs about that instance?

aignacio, 2021-06-24:
Sorry, I just check the report and it's not LUTRAM it's LUTs. About the logs I searched now and couldn't find any warnings due to that...
[image]
[rep_hier.zip](https://github.com/alexforencich/verilog-axi/files/6712287/rep_hier.zip)

alexforencich, 2021-06-24:
Well, I need to see the synthesis report.  Did you try changing PIPELINE_OUTPUT to 2 or 3 to see if that changes anything?

aignacio, 2021-06-24:
Here's the synthesis log, I'll try to change the parameters as you mentioned to see if It changes something, tks!
[runme.log.zip](https://github.com/alexforencich/verilog-axi/files/6712312/runme.log.zip)

alexforencich, 2021-06-24:
> INFO: [Synth 8-6085] Hierarchical reference on 'mem' stops possible memory inference [/home/aignacio/mpsoc_example/build/example_mpsoc_1.0.0_0/src/example_mpsoc_1.0.0_0/rtl/verilog-axi/rtl/axi_ram.v:145]

aignacio, 2021-06-24:
Interesting, I was looking for the wrapper name that's why I couldn't find such info, thx! The only hier. mention that I have is due to a Verilog function that's called by Verilator to load the elf on the memories, it seems odd that vivado takes it as part of synthesis, I'll remove that to see if it's that's why, anyway tks again!

alexforencich, 2021-06-24:
this is where the synthesis translate on/off directives can be helpful

aignacio, 2021-06-24:
I'm going to re-run with the directives, tks!

aignacio, 2021-06-25:
worked like charm, all rams converted into BRAMs, tks!

closed 2021-06-25 by aignacio