alexforencich/verilog-axi #18: How to get waves from the tbs

aignacio, 2021-07-21:
Hey @alexforencich,

how can I run a specific test with waves dump enabled? For instance axi_dma...
```bash
$ tox -- -k "axi_dma" WAVES=1 
```

alexforencich, 2021-07-21:
Not sure offhand how to do it with tox, but if you can set the WAVES env var inside tox I think it will work.  Usually when I want waves, I use the makefile flow to run the simulator.  At some point I am going to rework the python scripts so you can run the python testbench script directly and use a command line arg to enable waveform dumping, but that hasn't been implemented yet.

aignacio, 2021-07-22:
hmmm, I tried to run `make` in the `tb/axi_dma` and I had this error:
```python
/Applications/Xcode.app/Contents/Developer/usr/bin/make results.xml
MODULE=test_axi_dma TESTCASE= TOPLEVEL=axi_dma TOPLEVEL_LANG=verilog \
        /usr/local/bin/vvp -M /usr/local/lib/python3.9/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp -fst
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:77   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:240  in _embed_sim_init                 Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:212  in _initialise_testbench           Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     cocotb                                      __init__.py:218  in _initialise_testbench           Running tests with cocotb v1.5.0.dev243+g5d2d7c7c from /usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb
     0.00ns INFO     cocotb                                      __init__.py:239  in _initialise_testbench           Seeding Python random module with 1626978975
     0.00ns CRITICAL cocotb.regression                         regression.py:177  in _discover_tests                 Failed to import module test_axi_dma: cannot import name 'AxiBus' from 'cocotbext.axi' (/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotbext/axi/__init__.py)
     0.00ns INFO     cocotb.regression                         regression.py:178  in _discover_tests                 MODULE variable was "test_axi_dma"
     0.00ns INFO     cocotb.regression                         regression.py:179  in _discover_tests                 Traceback:
     0.00ns INFO     cocotb.regression                         regression.py:180  in _discover_tests                 Traceback (most recent call last):
                                                                                                                       File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/regression.py", line 175, in _discover_tests
                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/regression.py", line 66, in _my_import
                                                                                                                         mod = __import__(name)
                                                                                                                       File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/_pytest/assertion/rewrite.py", line 170, in exec_module
                                                                                                                         exec(co, module.__dict__)
                                                                                                                       File "/Users/aignacio/projects/verilog-axi/tb/axi_dma/test_axi_dma.py", line 38, in <module>
                                                                                                                         from cocotbext.axi import AxiBus, AxiRam
                                                                                                                     ImportError: cannot import name 'AxiBus' from 'cocotbext.axi' (/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotbext/axi/__init__.py)

Traceback (most recent call last):
  File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/__init__.py", line 274, in _initialise_testbench
    regression_manager = RegressionManager.from_discovery(top)
  File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/regression.py", line 154, in from_discovery
    return cls(dut, tests, hooks)
  File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/regression.py", line 126, in __init__
    for test in tests:
  File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/regression.py", line 175, in _discover_tests
    module = _my_import(module_name)
  File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/cocotb/regression.py", line 66, in _my_import
    mod = __import__(name)
  File "/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/_pytest/assertion/rewrite.py", line 170, in exec_module
    exec(co, module.__dict__)
  File "/Users/aignacio/projects/verilog-axi/tb/axi_dma/test_axi_dma.py", line 38, in <module>
    from cocotbext.axi import AxiBus, AxiRam
[... cut ...]
     0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:324  in _embed_sim_init                 cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:295  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
ERROR: results.xml was not written by the simulation!
make[1]: *** [results.xml] Error 1
make: *** [sim] Error 2
```
do you know why? 

my version:
```bash
cocotb               1.5.0.dev243+g5d2d7c7c
cocotb-bus           0.1.1
cocotb-test          0.1.1
cocotbext-axi        0.1.4
```

alexforencich, 2021-07-22:
> cocotbext-axi        0.1.4

time to upgrade

aignacio, 2021-07-23:
true =) tks

closed 2021-07-23 by aignacio