alexforencich/verilog-axi #19: Correct understanding of axi_dma

aignacio, 2021-07-30:
Hey @alexforencich,

thanks for the IPs, such amazing job for the open source community! I have a question for you about the `axi_dma` design, basically from my understanding the write descriptors interface (`s_axis_write_desc_*`) configures the way the `s_axis_write_data*` will be reflected/processed in the m_axi_aw/w channels so if you see the image below, in this test, the descriptor length is 7 bytes. I see that in the write address channel you launch a two beat burst due to fact you can only send 4 bytes per beat. Although in the write data channel I see that you only send 5 bytes, is this because the randomisation of the test didn't set the other 2x tkeep bits and you just mirror the same in the write data channel or am I missing something? why the descriptor finishes with only 5 bytes transferred?

[image]

alexforencich, 2021-07-30:
The transfer length in the descriptor is basically the maximum transfer size.  The DMA engine will write up to that number of bytes from the AXI stream packet, and discard any overage.  So in that test case, the requested transfer length was 7, but the AXI stream packet was only 5 bytes long, so it only wrote 5 bytes, and reported that it wrote 5 bytes.

aignacio, 2021-07-31:
I see, gotcha, tks again!

closed 2021-07-31 by aignacio