alexforencich/verilog-axi #20: questions about data transfer order

Hang-XX, 2021-08-02:
Hi @alexforencich   
    when i read the file axi_cdma.v , I have a question about the data transfer order. when output is not ready , data will be stored int the temp. When the next cycle output is ready, the data in temp should be sent to the output first. But in line 773, int_to_output has the highest priority, and the input data is sent directly to output, skipping the data in temp.
    The reg m_axi_wready_int_reg is only associated with input status，so I wonder if there's a problem with the logic here.

(line 773,axi_cdma.v)
`    if (m_axi_wready_int_reg) begin
        // input is ready
        if (m_axi_wready || !m_axi_wvalid_reg) begin
            // output is ready or currently not valid, transfer data to output
            m_axi_wvalid_next = m_axi_wvalid_int;
            store_axi_w_int_to_output = 1'b1;
        end else begin
            // output is not ready, store input in temp
            temp_m_axi_wvalid_next = m_axi_wvalid_int;
            store_axi_w_int_to_temp = 1'b1;
        end
    end else if (m_axi_wready) begin
        // input is not ready, but output is ready
        m_axi_wvalid_next = temp_m_axi_wvalid_reg;
        temp_m_axi_wvalid_next = 1'b0;
        store_axi_w_temp_to_output = 1'b1;
    end
`

alexforencich, 2021-08-02:
Are you running in to a bug, or just trying to understand the logic?  I use the same output logic on a gigantic number of modules, and I have not run in to any issues with it in its current incarnation.  But it is old code that could possibly be cleaned up a bit.  

Anyway, `m_axi_wready_int_reg` follows `m_axi_wready` with a delay of one cycle, so when `m_axi_wready` rises `store_axi_w_temp_to_output` will be set first, transferring the temp register to the output.  On the next cycle, the temp register will be empty.

Hang-XX, 2021-08-03:
I got it. I ran into a bug because i changed 'temp' into a fifo, so the logic here needs to be modified, there is nothing wrong with your logic.
Thanks for your help!

alexforencich, 2021-08-03:
Yeah, this module is not super optimized because I have actually never used it for any actual designs.  A shallow output FIFO would certainly be a good idea, probably for all of the DMA modules.  I'll put adding output FIFOs to my to-do list.

alexforencich, 2021-08-03:
You know what, I have some output FIFO code in some of the PCIe DMA modules that I can just paste in.  Does this help at all for your application?

https://github.com/alexforencich/verilog-axi/commit/849addb98fb0b33d5920d42360e6f4a203ca0c69

Output FIFOs of 32 entries, optimized for efficient implementation on distributed RAM, with half-full used as early backpressure.

Hang-XX, 2021-08-03:
Thank you! It really helps a lot!