alexforencich/verilog-axi #21: questions about axicrossbar

omeag, 2021-08-10:
hey sir :
I just start learning axi stuff, when i read pg059-axi-interconnect [xilinx], im confused about 'Avoiding Deadlock Using Single Slave Per ID', i want to know if you implement this function in axi crossbar? and what part of the code achieved that?
Thanks a lot.

alexforencich, 2021-08-10:
That's tracked here:

https://github.com/alexforencich/verilog-axi/blob/master/rtl/axi_crossbar_addr.v#L236-L269

omeag, 2021-08-11:
Thank you! I'll read this carefully. In addition, I was wondering if this axi-Crossbar supports read-only or write-only devices ? 
and, I want to add axi-fifo in to the axi-crossbar. How can i implement it ? Is it similar to the axi-register？

omeag, 2021-08-13:
hi ,sir . I've got another question when I simulate the data-width module.  when a master read data from a slave, i got this in MI:
[image]
but SI:
[image]
[image]
[image]

i'm wondering why i got this? (SI data width = 32, MI data width = 16)
Looking forward to your reply. TKS

omeag, 2021-08-13:
plus, i want to add data-width module in to the crossbar, i don't know how to do it...could you please help me?

alexforencich, 2021-08-14:
That looks fine to me.  The value of rdata is irrelevant when rvalid is low, only the middle trace shows rdata coincident with rvalid high.  

The crossbar doesn't have provisions to include the data-width converter in the same module.  You'll have to wire it up in series with whatever port(s) you want it on.

omeag, 2021-08-16:
Thank you! sorry to bother you but.. Im not clear about the meaning of 'S_THREADS' , 'M_ISSUE' and 'S_ACCEPT'.
if there are three transactions recived on SI: 
trans 0     |    trans 1   |   trans 2   |
arid = 0   |    arid = 1 |    arid = 0 |
S_THREADS = 2, S_ACCEPT = 3.  Is this right?  and what should the value of 'M_ISSUE' be?

omeag, 2021-08-19:
HI, sir 
when I simulate the data-width module, i found it always missing the first data

i set the data width of SI is 64, MI is 32. and the burst start at 1111_2222_7700_77aa

but i got this: the first data is 0000_0001, is there something i didn't notice? 

And why the 'valid' turn to low in the middle? 
Looking forward to your reply. TKS
[image]

alexforencich, 2021-08-19:
I need to see the traffic on both s_axi_* and m_axi_*

omeag, 2021-08-19:
that is:
[image]

alexforencich, 2021-08-19:
the first cycle of the transfer on the SI is 0000000000000001, so that gets split into 00000001 and then 00000000, then you get 11112222770077ab that gets split into 77077ab then 11112222, etc.  Looks fine to me.

omeag, 2021-08-19:
And this is my master:
[image]

but when i ues the axi_datawidth_converter from vivado ip catalog,
the same master :
[image]

alexforencich, 2021-08-19:
There is an AXI protocol violation in that second trace.   wdata cannot change when wvalid = 1 but wready = 0.  It must hold the value 0000000000000001 until wready = 1.

omeag, 2021-08-19:
thank you ! But how can i prevent 0000_0001 being split as the first data, since i didn't transmit it..

alexforencich, 2021-08-19:
You did transmit it as you set wvalid high when wdata carries the value 0000000000000001.  Perhaps you're setting wvalid high one cycle too early; you should only set wvalid high when you present the value you want on wdata.

omeag, 2021-08-19:
i see, I'll try to adjust it. thank you ! 
I also wonder why in this pic, 'wvalid' turn to low while 'wdata' have a duplicate data?
[image]

alexforencich, 2021-08-19:
it goes low because the next block of data is not yet ready to transfer out.  Previously it transferred 77077ab, then 11112222, then it goes back to 770077ab with wvalid low (so no transfer takes place and that data is ignored), then it outputs "new" data (77077ab).  And since this is new data, it sets wvalid = 1 to transfer it.  

Possibly there are some optimizations to be made to remove some of those cycles with wvalid = 0.  This module TBH has a few performance issues that I may be looking at in the next few months.

omeag, 2021-08-19:
extremely thanks your explaining !

omeag, 2021-08-24:
Hi, sir
Do you plan to add clock domain crossing for axi-full ? I want to add it in axi-crossbar.

alexforencich, 2021-08-25:
It's on my to-do list as part of some of the new capabilities of Corundum; sometime in the next few months.  Most likely coupled with re-working of the width converter.

omeag, 2021-08-25:
OK ! thanks

omeag, 2021-09-23:
Hi, sir
I have tried to write CDC module, and I understand how to write async FIFO(using gray code), but I don't know how to make axi-full interface.  For example, how should ID signals and address signals be saved in async FIFO? 
And may I take the liberty to ask if you could write axi-full cdc first?   ToT