alexforencich/verilog-axi #22: application problem

tuuyii, 2021-09-04:
Hi, Alex
Can Axi-interconnect be used in ASIC? And I want to apply it to a school teaching project.
Thanks,
Tuuyii

alexforencich, 2021-09-04:
I don't see any reason why not.  I avoid FPGA-specific primitives as much as possible as this makes the code more portable, but at the same time I tend to use quite a few inferred RAMs.  This works fine on FPGAs, ending up in block RAM or distributed RAM, but presumably on an ASIC this would end up inferring a large number of flip-flops.  Which works from a functional perspective, but is not really ideal.

mithro, 2021-09-06:
@alexforencich - Not technically an AXI-Interconnect issue, but it would be awesome if we could use Yosys to do "auto inference" of memory blocks in ASIC flow in a similar manner to how these designs work in FPGAs land. 

If someone like @tuuyii is interested in that, I would suggest first getting the designs to work with open source FPGA tool suites like SymbiFlow which uses Yosys internally. The excellent work that @mwkmwkmwk has been doing to improve memory block support in Yosys also makes this idea more feasible too.

Once a design can go to an FPGA using only open source tools, it then becomes much easier to start porting to the open source ASIC toolchains like OpenROAD / OpenLane. I know that both @The-OpenROAD-Project and @antmicro  are looking into things like this to make it easier to use things like open source FPGA targeted designs in the open source ASIC flows.