alexforencich/verilog-axi #23: About CDC module

omeag, 2021-09-08:
Hi, sir
I have some questions about the synchronization techniques. 
1. Is your axil-CDC implemented with dual latches? Or MCP formulation? 

2. I want to implement a axi-full cdc module. so, for AXI-Full , is clock domain synchronization implemented using an asynchronous FIFO? 

Sorry to bother you. ^ ^. I'm a little weak on the basics, so I'm hoping for your help badly! Thank you!
Best wishes.

alexforencich, 2021-09-08:
It's a simple handshake synchronizer.  An operation is accepted on the slave interface, the request is synchronized across with a pulse synchronizer and presented on the master interface, then the response is accepted, synchronized back with a pulse synchronizer and presented on the slave interface, and then a new operation is accepted.  Only one operation can take place at a time.  It's not very high performance, but it works great for simple register interfaces.  It's also specifically designed such that either side can be reset at any time without breaking the AXI lite protocol (although you may get a false read or write on the M side).

For AXI, it would make more sense to use async FIFOs.  Implementing that is on my to-do list in the next couple of months, along with possibly reworking some of the other modules.

omeag, 2021-09-09:
Thank you for your detailed answer！It helps a lot.  ^v^
And here's another question: if the clk of S side is faster than that of M side and have two transactions closed, will one transaction be lost? Is it necessary to use async FIFOs?

alexforencich, 2021-09-10:
No, because it will block any new transactions until the current one completes.  It's a very simple CDC block that's designed to gracefully handle one side or the other being reset.  For full AXI, I would recommend using full async FIFOs and then making sure you always reset the whole design at once.