alexforencich/verilog-axi #29: Question on address decoder to determine master interface

Pi-Turn, 2021-12-23:
I took some time to understand the code below and has the following question

`            for (i = 0; i < M_COUNT; i = i + 1) begin
                for (j = 0; j < M_REGIONS; j = j + 1) begin
                    if (M_ADDR_WIDTH[(i*M_REGIONS+j)*32 +: 32] && (!M_SECURE[i] || !axi_prot_reg[1]) && ((read ? M_CONNECT_READ : M_CONNECT_WRITE) & (1 << (s_select+i*S_COUNT))) && (axi_addr_reg >> M_ADDR_WIDTH[(i*M_REGIONS+j)*32 +: 32]) == (M_BASE_ADDR_INT[(i*M_REGIONS+j)*ADDR_WIDTH +: ADDR_WIDTH] >> M_ADDR_WIDTH[(i*M_REGIONS+j)*32 +: 32])) begin
                        m_select_next = i;
                        axi_region_next = j;
                        match = 1'b1;
                    end
                end
            end`

May I know why it is necessary to have a long if condition list? For example...

`M_ADDR_WIDTH[(i*M_REGIONS+j)*32 +: 32]` <==This will be always TRUE no matter how as long as user define those parameters.

`((read ? M_CONNECT_READ : M_CONNECT_WRITE)` <== This will be always TRUE as well.

I might be wrong in my interpretation.
If you don't mind, could you please share probably they is a strong reason to have this checking?

alexforencich, 2021-12-23:
Your first assumption is incorrect, `M_ADDR_WIDTH[(i*M_REGIONS+j)*32 +: 32]` will be zero if M_ADDR_WIDTH for that port and region is set to zero, indicating that it is not used.  Also, `(read ? M_CONNECT_READ : M_CONNECT_WRITE)`  is not used by itself, that is bitwise ANDed with something else: `((read ? M_CONNECT_READ : M_CONNECT_WRITE) & (1 << (s_select+i*S_COUNT)))`.

Pi-Turn, 2021-12-23:
Thanks for the prompt response. I agree with you .

I was referring to the default parameter setting below, but I forgot when user override it from top level, certain port can be entirely disable by setting those port parameter to 0. And the same applied to M_CONNECT_READ & M_CONNECT_WRITE.
` parameter M_ADDR_WIDTH = {M_COUNT{{M_REGIONS{32'd24}}}},`

alexforencich, 2021-12-23:
Yep, the whole idea is the user can either keep the default parameter values, or override them as necessary directly in the RTL, as opposed to generated cores where this sort of per-instance overriding is not possible.

Pi-Turn, 2021-12-23:
Another thing that I curious about is, it seems to me the AXI multiple signalling support is quite straight-forward in your implementation. But surprisingly, Xilinx does not support it in their interconnect IP based on the 2021 user guide.

[https://support.xilinx.com/s/question/0D52E00006hpYue/arawregion-missing-on-slave-ports-of-crossbar?language=en_US](url)

Any speculation on what lead to they don't support this?

alexforencich, 2021-12-23:
No clue, possibly because none of their other IP needs it.  I also haven't needed it, but I just figured I might as well implement it, and it will be optimized out if the region signals are not used.

Pi-Turn, 2021-12-23:
Thanks for replying the question & the efforts of the work! Really appreciate.
I will mark this to "close" status now.

closed 2021-12-23 by Pi-Turn