alexforencich/verilog-axi #31: AXI interconnect hanging while mult. writes

aignacio, 2022-02-22:
Hey @alexforencich,

I'm using your _axi_interconnect_ design to build a small SoC and I'm having an issue where a CPU is hanging due to the intcon not being available to receive more requests. The configuration for the interconnect is:

```verilog
axi_interconnect_wrapper #(
    .N_MASTERS        (2),
    .N_SLAVES         (4),
    .M_BASE_ADDR      ({32'hB000_0000, 32'hA000_0000, 32'h8000_0000, 32'h1000_0000}),
    .M_ADDR_WIDTH     ({32'd17, 32'd17, 32'd17, 32'd17})
  ) u_axi_intcon (
    .clk              (clk),
    .arst             (rst_int),
    .*
  );

```
Basically M0 is CPU fetch I/F (only rd requests) and M1 is CPU I/F LSU, and the 4x slaves are AXI memories.  I'm attaching the waves and a template for gtkwave if you can take a look and have a better guess but it's not that clear what can lead to such scenario. Also, is there a way/config to improve the interconnect throughput?

[image]

[Archive.zip](https://github.com/alexforencich/verilog-axi/files/8118941/Archive.zip)

alexforencich, 2022-02-22:
The AXI interconnect block is a very simple block that only supports a single concurrent operation.  If you want higher throughput, use the AXI crossbar, which supports multiple concurrent operations.

As for the hang, you didn't set wlast correctly.  Should be high on the last cycle of the burst.  So the interconnect assumes that the burst has not yet finished, and it blocks waiting for more data.

aignacio, 2022-02-23:
@alexforencich,

After fixing the wlast the intcon **started working perfectly**. So I replaced the intcon with the same config by the crossbar as you suggested and now I'm seeing some hangs but this time is not due to the wlast, it seems something due to the number of outstanding operations I guess. Is there any sort of limitation on that? I'm attached the waves of a simulation with the crossbar...

[cross_hang.zip](https://github.com/alexforencich/verilog-axi/files/8128562/cross_hang.zip)

aignacio, 2022-03-05:
@alexforencich what are the restrictions of usage for the axi_crossbar?