alexforencich/verilog-axi #32: About CDMA testbench with AXI_DATA_WIDTH > 64

swleungbrian, 2022-03-08:
Dear Sir,

I am trying to simulate with the testbench cocotb code with AXI_DATA_WIDTH above 64. 
It went fine with 32 and 64 but failed when I set data width to 128. Not sure if the verilog module does not support it or I need to revise the Makefile and/or the test_axi_cdma.py ?

[image]

Thanks,

Brian

alexforencich, 2022-03-08:
Looks like it's a testbench issue.  Changing

    range(8, 8+byte_lanes*2, step_size)

to

    range(16, 16+byte_lanes*2, step_size)

enables the testbench to run.  So, it looks like I just need to revisit some of the testbench parametrization.

swleungbrian, 2022-03-09:
it works. thanks

closed 2022-03-09 by swleungbrian