alexforencich/verilog-axi #35: question: how to read this syntax?

hughperkins, 2022-05-10:
Hi, this isnt an issue, it's a question, about verilog, since you're using a notation I haven't seen before, and not quite sure how to Google.

At e.g. https://github.com/alexforencich/verilog-axi/blob/25912d48fec2abbf3565bbefe402c1cff99fe470/rtl/axi_fifo_wr.v#L183

```
reg [7:0] m_axi_awlen_reg = 8'd0, m_axi_awlen_next;
```

So, I'm interpreting this to mean that, you create an 8 bit register `m_axi_awlen_reg`, and assign i the value of `m_axi_awlen_next`. However, I'm not sure what the `8'd0,` does? I'm guessing it does something like assert the number of vits in `m_axi_awlen_next`? (I looked through verilog-std-1364-2005.pdf, eg section 6.2, 9.2, but wasn't able to find the relevant syntax).

alexforencich, 2022-05-10:
Declare 8 bit reg `m_axi_awlen_reg`, initialize it to zero, declare 8 bit reg `m_axi_awlen_next`, initialize it to `X`.

hughperkins, 2022-05-10:
Oh.... thanks! I assumed it was a declaration of a single register, but I see now that it declares too. Got it. thanks! :)

closed 2022-05-10 by hughperkins