alexforencich/verilog-axi #38: Parametrization doesn't work correctly in testbenches

vkomenda, 2022-08-26:
The `data_width` parameter is constant 32 in the tests according to logs, despite it having been defined to take values from the `[8, 16, 32]` range.

```
 90890.02ns INFO     AXI lite master configuration:
 90890.02ns INFO       Address width: 32 bits
 90890.02ns INFO       Byte size: 8 bits
 90890.02ns INFO       Data width: 32 bits (4 bytes)
 90890.02ns INFO     AXI lite master signals:
 90890.02ns INFO       awaddr width: 32 bits
 90890.02ns INFO       awprot width: 3 bits
 90890.02ns INFO       awready width: 1 bits
 90890.02ns INFO       awvalid width: 1 bits
 90890.02ns INFO       wdata width: 32 bits
 90890.02ns INFO       wready width: 1 bits
 90890.02ns INFO       wstrb width: 4 bits
 90890.02ns INFO       wvalid width: 1 bits
 90890.02ns INFO       bready width: 1 bits
 90890.02ns INFO       bresp width: 2 bits
 90890.02ns INFO       bvalid width: 1 bits
```

Logs for all tests show 32-bit data width.

vkomenda, 2022-08-26:
Running tests with `pytest` instead of `make` fixes that.

closed 2022-08-26 by vkomenda

alexforencich, 2022-08-27:
The jist of it is this: parameters (generally) cannot be changed during the simulation, so the simulation must be re-run separately for each set of parameters.  The makefile flow only runs one set of parameters, but you can easily override parameters on the command line for testing purposes.  The pytest flow uses pytest.mark.parametrize to generate a set of tests that cover a range of parameter settings.  The github actions CI simply runs pytest to run all of the parametrized tests.