alexforencich/verilog-axi #44: AXIL crossbar doesn't support M_ADDR_WIDTH < 12

martin-tanguay, 2023-02-13:
Hi,

I've checked out you "verilog-axi" repo (which is really cool and contains pretty much all the AXI tools) and I've found a bug in the AXIL crossbar where it doesn't support M_ADDR_WIDTH < 12.  I've created a testbench with one master and 2 slaves and it works perfectly for M_ADDR_WIDTH >= 12, but asserts an error for M_ADDR_WIDTH < 12.

The root cause is in axil_register_wr.v (line 131), in axil_register_rd.v (line 119) and in axil_crossbar_addr.v (line 138).  Looking at the AXIL interconnect, I've found that this limitation doesn't exist, so I copied the address check in the AXIL crossbar (M_ADDR_WIDTH[i*32 +: 32] < 0) and it works perfectly fine.

Is there any reason for not supporting M_ADDR_WIDTH < 12 ?

By the way, the AXI crossbar (not AXIL) also has the same limitation.

Thanks.

Martin Tanguay

martin-tanguay, 2023-02-13:
Thanks for the quick reply.  I understand that in the AXI spec, bursts cannot cross 4096 byte boundaries, but can be less than that.
A good example would be a slave module with only 64 data addresses (6 bits of address), why would we lose 12 bits of address space ?

alexforencich, 2023-02-13:
You wanna rewrite the crossbar to deal with splitting bursts across multiple ports?  If you're slicing and dicing the address space on such small granularity, it probably makes sense to use AXI lite anyway instead of AXI.  Or perhaps even AHB.

martin-tanguay, 2023-02-13:
Understood.  And by the way, I really didn't meant that the crossbar needs to be rewritten. On the contrary, the suite is really great, and I wanted to share if that could be an limitation...

Thanks for your answers.

alexforencich, 2023-02-13:
No worries; and thanks for reporting this for the AXI lite crossbar, as this restriction isn't required for AXI lite as AXI lite does not support bursts.  

For AXI, you're normally only going to use that when you need to move around lots of data with bursts, perhaps interfacing with things like DRAM controllers.  In which case all of the address ranges are going to be much larger than 12 bits.  For lots of small peripherals, it makes a lot more sense to use a simpler protocol like AXI lite, AHB, wishbone, etc. which doesn't require all of the additional logic to support bursts.  You can dedicate one larger address range on the AXI crossbar, then adapt that to AXI lite and set up an AXI lite crossbar for all of the peripherals.

alexforencich, 2023-02-13:
Also, I just pushed a new version with this fix, please let me know if it works for you.

martin-tanguay, 2023-02-14:
Thanks Alex, I appreciate.  Great set of tools !  I'm looking forward to use them.  Martin