alexforencich/verilog-axi #59: about AXI_VFIFO

Monster-Kee, 2023-09-24:
I tried to use the AXI_vfifo module you wrote, and I tried to configure it as a single channel FIFO. The other side is connected to a DDR MIG controller, it seems that the input of the FIFO is normal, and the AXI reading and writing of the MIG are also normal, but the FIFO has no output. I don't know where the problem lies at the moment

ThomasVec, 2023-11-27:
I have the same problem with the simulation of the vFIFO. It seems that the decoding module does not start decoding. The data from the vFIFO rd goes to the decoding module, with a valid signal. The decoding module does not terminate the data with Ready.
[image]

Used Parameters for vFIFO:
      -- AXI channel count
        axi_ch             => 1,
      -- Width of AXI data bus in bits
        axi_data_width      => 32,
      -- Width of AXI address bus in bits
        axi_addr_width      => 32,
      -- Width of AXI wstrb (width of data bus in words)
        axi_strb_width      => 32/8,
      -- Width of AXI ID signal
        axi_id_width        => 3,
      -- Maximum AXI burst length to generate
        axi_max_burst_len   => 256,
      -- Width of AXI stream interfaces in bits
        axis_data_width     => 32,
      -- Use AXI stream tkeep signal
        axis_keep_enable    => 1,
      -- AXI stream tkeep signal width (words per cycle)
        axis_keep_width     => 32/8,
      -- Use AXI stream tlast signal
        axis_last_enable    => 1,
      -- Propagate AXI stream tid signal
        axis_id_enable      => 1,
      -- AXI stream tid signal width
        axis_id_width       => 3,
      -- Propagate AXI stream tdest signal
        axis_dest_enable    => 1,
      -- AXI stream tdest signal width
        axis_dest_width     => 3,
      -- Propagate AXI stream tuser signal
        axis_user_enable    => 1,
      -- AXI stream tuser signal width
        axis_user_width     => 1,
      -- Width of length field
        len_width           => 8,
      -- Maximum segment width
        max_seg_width       => 256,
      -- Input FIFO depth for AXI write data (full-width words)
        write_fifo_depth    => 64,
      -- Max AXI write burst length
        write_max_burst_len => 16,
      -- Output FIFO depth for AXI read data (full-width words)
        read_fifo_depth     => 128,
      -- Max AXI read burst length
        read_max_burst_len  => 16

ivansun1688, 2024-03-06:
Hi,

currently I set axi_data_width=256, ch = 1, and then set axis_data_width=axi_data_width/2, max_seg_width = axi_data_width/2, and then the vfifo works.

if axi_ch=1 then  set axis_data_width=axi_data_width and max_seg_width = axi_data_width, parameter SEG_CNT will to be 1, and parameters in encode/decode fifo are not correctly calculated and the vfifo will failed.

Thanks, please take a review.