alexforencich/verilog-axi #60: Q: Do axi_dma_rd and axi_dma_wr support out of order transactions?

abarajithan11, 2023-11-01:
I'm planning to use your DMAs in a high performance design. Since DDR delivers data after several clocks, it would be ideal if the DMA can issue multiple outstanding AXI transactions, without waiting for each transaction to complete.

Is this feature already included? Or are you planning to add it in the future?

Btw, thanks a lot for this amazing repo!!

alexforencich, 2023-11-02:
That should be supported.  However, I should note that read data interleaving is not supported.  I think these cores should use only ID 0 to prevent interleaving, but I will have to double check on that.  I don't think the Xilinx MIGs interleave, but I do know that the Zynq PS does interleave read data.  I'm planning on making sure everything in this repo supports read data interleaving where possible, but have not yet had the time to do so.

abarajithan11, 2023-11-08:
Thank you!

To clarify, the `axi_dma_rd` 
1. accepts a descriptor through the read descriptor channel
2. issues (burst?) AXI requests through the AXI master channel, delivering the data it gets through M_AXIS channel
3. AFTER completing the packet, i.e, after `m_axis_read_data_tlast` goes high, accepts the next descriptor.

Is my understanding correct? 

Or, does it accept the next descriptor **after issuing the last AXI request through AXI master**, and start issuing AXI requests for the next packet before `m_axis_read_data_tlast` of the first packet goes high?