alexforencich/verilog-axi #65: Will unprocessed awvalid signals be stored in AXI Crossbar?

omeag, 2024-01-23:
Hi, sir
I see that S_THREADS is set to 2 by default, which means that SI can initiate two outstanding transfers. 
So will the second awvalid and awaddr signal be stored? 
During simulation, I did not keep awvalid high all the time. The results show that it will not process the second write operation. 
How does your AXI crossbar handle outstanding transfers?

alexforencich, 2024-01-23:
Well, you have to respect the handshake protocol.  *valid MUST be held high until *ready is asserted.  If you don't do that, things break in strange ways.

omeag, 2024-01-24:
So if I set S_THREADS to 3 and initiate three write operations, then I need to keep awvalid high until all three write operations are complete, right? 
But the write address signal will be staged? Will the second write address be lost?

alexforencich, 2024-01-24:
I recommend reading the AXI specification.

The short story is this: data is ONLY transferred when BOTH *ready and *valid are asserted at the same time.

This means that if *ready is low, the only thing you can do is set up a new operation (set all of the signals and set *valid = 1).  If *valid is already high, you must wait wait for *ready to be asserted.  If you change anything when *valid = 1 and *ready = 0, Bad Things will happen (lost transactions, unpredictable behavior, etc.).  Similarly, when both *ready and *valid are asserted, you must either deassert tvalid, or set up a new operation.  All of this follows from data being transferred only when both *ready and *valid are asserted at the same time.

The spec also states that you are not allowed to wait for *ready to be asserted before starting a new operation.  The sink is allowed to wait for *valid before asserting *ready, but the source is not allowed to do the converse as this could result in a deadlock.

omeag, 2024-01-24:
So what is the role of  OUTSTANDING?
As you said, it is necessary to keep awvalid and awaddr until awready is high and a successful handshake is done. 
However, the Outstanding function allows the Slave Interface initiate multiple operations while waiting for AWREADY, right? 

But if I additionally initiate multiple operations while waiting for AWREADY, I have to keep AWVALID and AWADDR ? So in this way, that means I can only initiate ONE another operation, and keep every signal until  AWREADY turns low.  

Where did I misunderstand?

alexforencich, 2024-01-24:
No, it means you can issue multiple AW transactions before they are completed on B.

omeag, 2024-01-24:
I understand completely now. Thank you so much for your explanation!