alexforencich/verilog-axi #66: Axi DMA consistently returns DECERR

EnricoGiordano1992, 2024-01-23:
Hello,

I am currently working on utilizing the Axi DMA module through PYNQ. I have created an HLS Axi Lite module that allows me to control the Axi DMA IP module using wires. However, whenever I attempt to send a stream of 128 32-bit values (setting the addressing to 32 bits in your module), I observe that the array of values I send is not being stored (the flow should be: I send an array of values via Axi Stream, and I expect them in the CPU's address space, received through AXI). Upon closer inspection, I noticed that with every transaction, I encounter the DECERR error.

The configuration flow I use for the Axi DMA module is as follows:

**Phase 1 (simultaneously):**
- Set `write_enable` to 1
- Set `write_abort` to 0
- Set `read_enable` to 0
- Set `s_axis_write_desc_addr` to the physical address of the array
- Set `s_axis_write_desc_len` to 128
- Set `s_axis_write_desc_tag` to 0

**Phase 2:**
- Set `s_axis_write_desc_valid` to 1

**Phase 3:**
- Send the values via stream

**Phase 4:**
- Read the values in the CPU (and observe all values as 0)

Particularly, when analyzing with ILA, I consistently observe the DECERR error. I would appreciate any guidance or insight into resolving this issue.

Thank you.

alexforencich, 2024-01-23:
Is this a spurious DECERR, or does this correspond to a DECERR on bresp?  If the latter, you have the destination address wrong or something else in your AXI infrastructure is screwed up.

EnricoGiordano1992, 2024-01-23:
I confirm that DECERR is on bresp.
[image]

Can you suggest to me how to check where could be the problem?
This is my Vivado design
[design_1.pdf](https://github.com/alexforencich/verilog-axi/files/14028106/design_1.pdf)

alexforencich, 2024-01-23:
Sorry, I'm not familiar with the IPI flow.  At any rate, DECERR is generated by the interconnect when it cannot decode the address.  So either you're providing the wrong address, or the interconnect addressing is not configured correctly.

EnricoGiordano1992, 2024-01-23:
Good point. I have done another test, connecting the Axi DMA to an Axi BRAM Controller. In this configuration, it seems that everything is ok, so is missing "something" in the connection between the AXI in PL and the AXI HP0 in the PS. I will keep you posted 👌

EnricoGiordano1992, 2024-01-26:
@alexforencich did you ever tried the Axi DMA connected to the PS on a custom design? If yes, can you give me any advice for the Vivado / bare metal solution?

alexforencich, 2024-01-26:
Not that specific module, but I have used the AXI DMA IF module on a Zynq as part of Corundum and it worked fine.  This was with Linux running on the PS, and it basically worked on the first try once I got petalinux to build.