alexforencich/verilog-axi #68: About width missmatch

a60626316, 2024-02-14:
Hi @alexforencich ,

Sorry to bother you, but I've encountered some issues while using axi_interconnect as follows:
axi_interconnect.v, line 485
Continuous assignment width mismatch
1 bit (lhs) versus 2 bits (rhs).
Source info: assign s_select = (grant_encoded >> 1);

However, I've checked the parameters I set as follows:
parameter S_COUNT = 2,
parameter CL_S_COUNT = $clog2(S_COUNT);
wire [CL_S_COUNT:0] grant_encoded;
wire [(CL_S_COUNT > 0 ? CL_S_COUNT-1 : 0):0] s_select;

In the case of S_COUNT = 2, CL_S_COUNT will be equal to 1, which causes grant_encoded to be 2 bits and s_select to be 1 bit. It seems that other variables have similar errors, but since their data widths all depend on the size of S_COUNT, I am unsure how to resolve this issue.

Thank you very much for your help!

alexforencich, 2024-02-14:
Not sure what the problem is.  If grant_encoded is 2 bits, then (grant_encoded >> 1) is one bit, which matches s_select.

a60626316, 2024-02-15:
My idea is that in a typical shift statement, the bit widths of s_select and grant_encoded should be equal. Additionally, after the shift, the other end should automatically be padded with zeros. So, you mean that after grant_encoded is right-shifted by 1 bit, its higher bit(grant_encoded[1]) will be zero, and s_select will equal grant_encoded[0]. However, since this width mismatch is a warning message popped up by the tool, I am not sure whether it will affect the computation or the logic synthesis. Thank you for your reply.