alexforencich/verilog-axi #71: Failed to run test for AXI RAM with DATA_WIDTH=64 and ADDR=64

ManjunathKalmath, 2024-03-04:
Greetings,

I am using your AXI RAM with modifications in DATA_WIDTH = 64 and ADDR_WIDTH = 64. In the test_axi_ram I included run_test_write with write_qword and read_qword. I am attaching the portion of run_test_write here. 

The first write and read is successful but the second one failed and I can see an **ERROR with an Exception raised by this forked coroutine.**

I am adding screenshots of waveforms for your reference. 

I would like to get your guidance on resolving this error.

Thank you!!   
[image]

[image]

[image]

alexforencich, 2024-03-04:
Please attach the waveform dump file instead of posting screenshots, as the screenshots do not contain all of the signals.

ManjunathKalmath, 2024-03-04:
Thanks, I am attaching AXI_RAM zip file here.
[AXI_RAM.zip](https://github.com/alexforencich/verilog-axi/files/14485443/AXI_RAM.zip)

alexforencich, 2024-03-04:
I think I know what's going on here.  First of all, you cannot set ADDR_WIDTH too high, as this module actually implements a RAM of size 2**ADDR_WIDTH bytes.  So if you set it too large, the simulator will crash due to your system not having enough RAM, and 2**64 is 16 exabytes.  Second, I think there may be a truncation to 32 bits that takes place somewhere in the simulator.  So apparently that 2**64 gets truncated to 0, and all accesses to locations other than 0 result in X, which cocotbext-axi doesn't handle nicely due to the screwed up BinaryValue object in cocotb.

My suggestion is to keep the ADDR_WIDTH setting to something reasonable.  You're not going to implement GB of SRAM on chip, so ADDR_WIDTH should never be set larger than maybe 30 (and even that is likely to cause problems with RAM use in the simulator).

ManjunathKalmath, 2024-03-05:
Thanks Alex!!

I changed ADDR_WIDTH = 8. Now, it is working as expected.

closed 2024-03-05 by ManjunathKalmath