alexforencich/verilog-axi #72: AXI_Register hangs when SIM=verilator

ManjunathKalmath, 2024-03-07:
Greetings Alex,

I am trying to run tests for AXI_Register by setting SIM=verilator.

Test hangs at 50ns.

I am using verilator version 5.008 as cocotb supports verilator version 5.006 and later.

Can you guide me how to fix this?

Thanks Alex!!

alexforencich, 2024-03-07:
Lowest supported version for cocotb-bus is 5.020.  And I think the cocotb docs are going to be updated soon to reflect that the lowest supported version is 5.022.  So, update verilator and try again.

davekeeshan, 2024-03-08:
Thank you so much for this, I have been having issues with verilator and cocotb and I haven't had time to drill down into the problem.

As to compiling verilator, as someone who usually has to work inside corporate walls and doesn't have access to sudo or docker solutions and will be on weirdly supported systems (most EDA machines are RH8 and there has been some dependency issues there over the past 2 years), I have been delighted to find this tool, [spack](https://github.com/spack/spack), which allows you to compile tools as a regular user and it compiles the dependencies as you go.

The tool was designed by lawerence livermore for HPC superciomputer work, but I started adding in or updating some of the open EDA tools, so icarus, yosys, verible and verilator.  I just got verilator 5.022 added last night and a sim I had hanging now runs.

You can also get all your tools compiled from [oss-cad-suite-build](https://github.com/YosysHQ/oss-cad-suite-build)  but I have had issues trying to get some of them to run on some of my work machines, so the direct compile method using spack is the way to go for me.

ManjunathKalmath, 2024-03-08:
I tried with verilator version 5.022.

I observe the same, the test hangs at 50ns.

closed 2024-03-15 by ManjunathKalmath