alexforencich/verilog-axi #74: About AXI_FULL_CDC

LZR1567, 2024-03-30:
HI,Please allow me to take up a moment of your time.
I'm developing with Vivado of Xilinx and using AXI_Clock_converter IP. Here is my configuration:
[image]
Now, I want to replace this IP core with your axil_cdc code.I have added some signals for the FULL interface and made simple modifications to the code for your LITE interface CDC. However, it was not successful in the end. Is there anything to pay attention to when changing LITE to Full?
My modified code has been uploaded to the attachment.
Thank you very much!
[axi_clock_converter_0.zip](https://github.com/alexforencich/verilog-axi/files/14809949/axi_clock_converter_0.zip)

alexforencich, 2024-03-30:
A CDC module for full AXI requires a different technique than the AXIL CDC block in this repo.  AXI lite has a nice 1:1 correspondence across all of the channels.  AXI does not as it supports bursts.  The design of the AXI lite CDC block requires everything to be single cycle, it doesn't work with bursts.  I'm not sure offhand the best way to implement bursts; likely this would require some kind of FIFO.

Also, the AXI lite CDC module is quite low performance (it only supports one operation at a time).  This is usually reasonable for register operations, but this could be a problem for AXI transfers that expect a higher throughput.  Using a full set of FIFOs would help with this, but then there are potentially going to be issues with resets.  The AXI lite CDC module is quite nice in that it will always follow the AXI protocol despite one end or both ends being reset.  Getting similar behavior out of AXI with full FIFOs is not trivial.

LZR1567, 2024-03-30:
Thank you for your answer, Please allow me to ask one more question.
If we don't use the IP core provided by Xilinx to implement CDC with AXI FULL interface, which CDC module design ideas can we learn from? For example, asynchronous FIFO, etc