alexforencich/verilog-axi #75: cocotb makefile

ManjunathKalmath, 2024-04-09:
Greetings Alex, 

In the cocotb, makefile, we mention the VERILOG_SOURCES indicating the verilog design files. 
This is convenient if we have less number of verilog design files. 

In general we use filelist which contains all design files. Further, the filelist taken as an argument for simulator. 

Do you think cocotb makefile can be added with these filelist?

I would like to know about this.

Thanks Alex!!

alexforencich, 2024-04-09:
Can you provide some more details on what these file list files are?  I suspect they're plain-text files with one file name per line, relative to the location of the file-list-file?  Can file-list-files also contain the names of more file-list-files?  Do the file-list-files have a consistent extension?  Are they supported by multiple tools?

In terms of adding support, I think there are a couple of things to consider.  First, it would be trivial to add whatever simulator arguments you want, that's not a big deal.  But, the problem is that, unless the makefile itself can interpret the file list files and detect changes in the listed files, then it won't be possible to run the build process for the simulation properly (only when files are changed).

So, I'm thinking perhaps it makes sense to figure out how to parse the file list files and then just automatically add their contents as explicit sources; this way the makefile will work correctly and there won't be issues with only certain simulators supporting file-list-files.

ManjunathKalmath, 2024-04-11:
Yes whatever you have mentioned about the filelist is right. Sometime flielist also contain other filelist. 

Typically industrial simulators(questa, VCS) supports this.

closed 2024-05-08 by ManjunathKalmath