alexforencich/verilog-axi #76: about AXI DMA

zengzhengqi0524, 2024-05-08:
Hi, When I use the dma_wr module, I found a bottleneck in the transmission bandwidth；I want to know that dma module whether support outstanding transmission or not，thanks！

alexforencich, 2024-05-09:
I need to double check; I think it supports multiple outstanding transfers, but it's also possible that it doesn't.  I don't currently use most of the AXI components at the moment, so I haven't done much to improve/refine them recently.

zengzhengqi0524, 2024-05-14:
[image]
Judging from my simulation results，It should support outstanding transmission；The bandwidth issues described previously may be due to other issues。Thank you very much for your prompt reply！