alexforencich/verilog-axi #77: Performing multiple beat transfer

ManjunathKalmath, 2024-05-08:
Greetings Alex,

I would like to perform multiple beat transfer of beat size 32Bytes. 

I found that when bytearray is filled with more than desired data it splits the axi write length.

I tried the following and expecting the first beat will be written at address 0x1000 and second, one will be calculated based on burst_type(here INCR burst type) so, second beat will be written at address 0x1032. But, I did not observe output that I was expecting. 

addr = 0x1000
test_data = bytearray(b'\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88')
await tb.axi_master.write(addr, test_data, size=size)

Can you help me to understand this?

Thanks Alex

alexforencich, 2024-05-08:
What are you observing?

ManjunathKalmath, 2024-05-08:
I observed that address is not incrementing from 0x1000 to 0x1032.

ManjunathKalmath, 2024-05-08:
Got it Alex.

I am performing a single AXI write burst consisting of 2 beats with beat size of 32Bytes. I am expecting to see that second beat will be written at 0x101F.

alexforencich, 2024-05-08:
Beat size 32 bytes, meaning wdata/rdata width is 256 bits?  The first "beat" will get written at 0x1000-0x101F, the second "beat" will get written at 0x1020-0x103F.  But, since these beats are part of the same burst, you'll only see 0x1000 on awaddr.

alexforencich, 2024-05-08:
(It's also possible I'm not understanding what you're seeing, as you haven't actually shown me any log output or waveforms)

ManjunathKalmath, 2024-05-08:
> Beat size 32 bytes, meaning wdata/rdata width is 256 bits? The first "beat" will get written at 0x1000-0x101F, the second "beat" will get written at 0x1020-0x103F. But, since these beats are part of the same burst, you'll only see 0x1000 on awaddr.

Yes exactly, I mean Beat size 32Bytes is wdata/rdata width is 256 bits.

ManjunathKalmath, 2024-05-08:
You are correct, I am able to see only 0x1000 on awaddr.
I share you the waveform tomorrow.

alexforencich, 2024-05-08:
Well, it sounds like maybe you don't understand how AXI works.  The address for each burst is transferred on a totally separate channel from the data, and there is only one "beat" on the awaddr/araddr bus for each burst, even if the burst consists of multiple "beats" on wdata/rdata.  So only the starting address is transferred in every burst, it's up to the destination to interpret the burst information (size, length, type) and generate the addresses locally if necessary.  This is both an advantage and a disadvantage of AXI, but the logic to do this is relatively simple.

Alternatively, you can effectively disable bursts by specifying max_burst_len=1 when creating the AXI master instance, this will result in all transfers getting split into single-cycle/single-"beat" bursts, where each burst will consist of a single "beat" on awaddr/araddr and a single "beat" on wdata/rdata.

ManjunathKalmath, 2024-05-08:
I understood completely now. I have basic understanding of AXI, now I am getting better at it. 
As you mentioned, I observed the same in the waveform that single beat on the awaddr/araadr and multiple beats on wdata/rdata.

By the way, as per the AXI spec, if awlen/arlen = 1 it will result in 2 beats on wdata/rdata respectively. 
In order, to have 2 beats on wdata/rdata should I set max_burst_len=1 or max_burst_len=2 ?
Can you guide me on this?

alexforencich, 2024-05-08:
awlen/arlen are limited to max_burst_len-1.  So if you want two beats, set max_burst_len=2.

ManjunathKalmath, 2024-05-08:
Ok Thanks a lot Alex!!
I am closing this issue.

ManjunathKalmath, 2024-05-09:
I have set max_burst_len =2 to observe two beats on wdata. But failed, I am adding code and screenshot for your reference.

Code : 

addr = 0x1000

test_data = bytearray(b'\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88\x11\x22\x33\x44\x55\x66\x77\x88')

await tb.axi_master.write(addr, test_data, size=size)

for i in range(100):
await RisingEdge(dut.clk) 

[image]

ManjunathKalmath, 2024-05-09:
Thanks again Alex!!

closed 2024-05-09 by ManjunathKalmath