alexforencich/verilog-axi #78: Q: DMA, desc_len VS tlast in axi_dma_wr

abarajithan11, 2024-05-09:
Seeking further clarification on [this behavior](https://github.com/alexforencich/verilog-axi/issues/19) 

If `len = 50`, and if axis stream packet only has 40 words (tlast raised after sending 40 bytes), does the DMA end the transaction?

In other words, 
can I tie descripter length to some arbitrarily high constant, and expect the DMA to write the correct number of bytes based on axis stream tlast and keep?

Btw, I'm forever grateful for your open source work!

alexforencich, 2024-05-09:
Yes.  In the current setup, the data is passed through directly instead of being buffered, so the burst can't be stopped early.  So, the "unused" part of the burst is sent with wstrb = 0, so nothing extra is written.  At some point, I might consider reworking the DMA engine to add an internal buffer to both break the flow control (streaming data input stall won't stall outgoing write data) as well as ensure that the bursts are actually the correct length so cycles aren't wasted on the AXI side.  But, I haven't had the motivation to do this as I don't do much with AXI at the moment.

abarajithan11, 2024-05-10:
Great, thank you!

abarajithan11, 2024-05-10:
Great, thank you!

abarajithan11, 2024-05-12:
Another quick question:

If I set the burst size to 10 bytes, if `len = 50`, and if axis stream packet only has 40 words (tlast raised after sending 40 bytes):

The DMA would send 4 bursts of 10 bytes each and complete the transaction right?
Or does it send 5 bursts, but set all strobe to 0 at the last burst??